jnaulet/OpenPicoRTOS #67: CONFIG_DEADLOCK_COUNT not defined in RPi Pico Demo Make

joe-eleksys, 2023-09-12:
Hello:
Working to try out OpenPicoRTOS. I downloaded the 1.9.0 ZIP, unpacked it, navigated to the raspberry-pico demo and did a make. It failed with the error being the CONFIG_DEADLOCK_COUNT was not configured. 
I see that it is defined in the generated file autoconf.h which is in turn referenced at the top of main.c.
 
Though not a newbie to Linux and C I am definitely not a guru! I am working to understand this make error.
Any help with this would be appreciated.
Thanks

jnaulet, 2023-09-12:
This is a mistake, the quick fix is to include <generated/autoconf.h> in the main.c file.
Originally, this file was force-included in the GCC command line, but it created problems with the static analysis. Other demos have been fixed but this one went through.

A more permanent fix would be to include it in picoRTOS.h but it's not a 100% satisfying solution (yet).

jnaulet, 2023-09-12:
Sorry I was in a hurry and didn't understand your message very well, can you copy-paste the failed build console here please ?

jnaulet, 2023-09-12:
An attempt at fixing this once and for all has been added to branch v1.9.x, you can check it out and try

closed 2023-09-24 by jnaulet